rovercode/rovercode #13: Text-to-Speech

aninternetof, 2016-09-09:
Make the rover say something using a text-to-speech engine (online or local)

ryanknutson, 2017-04-10:
I found these two articles about it [https://goo.gl/SrZ7Dl](https://goo.gl/SrZ7Dl) and [https://goo.gl/w4ybiJ](https://goo.gl/w4ybiJ). You would need a speaker to attach as well (something like [this](https://goo.gl/uX42kX) would work perfectly).

aninternetof, 2017-04-11:
Reading the articles now. I had demoed it once before using Google's cloud API, but I don't know if that's as open/free as it used to be. I think IBM Watson has a free tier too. But I'm down for local options, too.

Asking, that speaker is awesome -- haven't seen anything like it before.

ryanknutson, 2017-04-11:
It's too bad it's not available anymore, I searched around and that's the only one I could find. It's definately possible to build one like that though.

aninternetof, 2017-04-11:
Agreed. And if we want to build something that's too demanding to run on the rover, we can always host it on our AWS and make it a rovercode.com API.