rovercode/rovercode #72: As a teacher, I need to supply the rover with wifi credentials

aninternetof, 2016-11-30:
- Read QR code with the camera (generated by Internet site)
- Bluetooth
- Text file (generated by internet site) on a USB stick

ductape, 2018-05-23:
In regards to the QR code, here are some links that might be helpful for future reference. 

- Here is a [pure JS QR code generator](http://jeromeetienne.github.io/jquery-qrcode/) that could put Wi-Fi credentials into a QR code locally in the browser and never send credentials over the network for privacy. Then the rover could be _aimed_ at the monitor to read the credentials. This generator could be embedded in the [rovercode-web](https://github.com/rovercode/rovercode-web) app. 
- Python + [zbar](http://zbar.sourceforge.net/) example of [reading QR codes from webcam](https://github.com/cuicaihao/Webcam_QR_Detector)
  - *Note*: [zbar](http://zbar.sourceforge.net/) seems unmaintained, and might not be a good choice to build around-- the latest commit seems to be [about six years ago](https://github.com/ZBar/ZBar), but I don't see any examples of anybody using anything else...
- [Raspberry Pi example](https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md) of setting up Wi-Fi through command line.

jeeshofone, 2018-05-23:
I too have looked any nearly everything seems to be zbar. I did find this: https://github.com/yushulx/opencv-python-webcam-barcode-reader/tree/master/raspberrypi
https://github.com/IdreesInc/QR-Code-Extractor
that don't use zbar but is this worth while to maintain ourselves?

ductape, 2018-05-23:
Currently, zbar has no [known vulnerabilities](https://www.cvedetails.com/google-search-results.php?q=zbar&sa=Search), so it seems _safe-ish_ to use. I haven't tested it yet, but if it does work, having no known vulnerabilities seems like the minimum bar a library should pass to be used in rovercode 😛

aninternetof, 2019-06-10:
Assigned to @jeeshofone. Let's start with a USB-stick solution, then move to the webcam+QR solution later?

jeeshofone, 2019-06-10:
Yes, I think that makes sense!