rovercode/rovercode #78: Make rover start on boot when on rover hardware

aninternetof, 2016-11-30:
But not on development hardware

ryanknutson, 2017-04-05:
Would adding the script to /etc/rc.local work? If not, a cron job would work well.

aninternetof, 2017-04-05:
Either would work great! I think I prefer the cron job.

I wonder if we can add something to setup.sh to create the cron job during install.

ryanknutson, 2017-04-05:
This command adds the script to /etc/rc.local
`sed -e '$i \sh /install/dir/start.sh \n' /etc/rc.local`

aninternetof, 2017-04-05:
I like this plan. Do things started through rc.local run with root privileges? start.sh will need them.

ryanknutson, 2017-04-05:
Yes, it runs as root.

aninternetof, 2017-04-05:
Next decision: how should we make setup.sh do this local.rc addition when installing on the target hardware (CHIP, RaspPi, BeagleBone), but not when installing on a development PC? 

We could have it try to read the cpu procinfo and figure out on its own what kind of system it is on. 

Or, we could just have the user pass a flag into setup.sh. That would probably be simpler, and less risk of it getting it wrong.

cabarnes, 2017-04-05:
You could have `setup.sh` ask the user rather than needing them to supply a parameter.

aninternetof, 2017-04-05:
I like that.

ryanknutson, 2017-04-05:
It should ask the user, that way a developer that isn't paying attention doesn't risk messing up their rc.local

ryanknutson, 2017-04-05:
I think I have it working. See f54d4eab3dcd0a6fdb5e8d452f1f15ccbbfd1ddb.

aninternetof, 2017-04-05:
LGTM. Want to make a pull request for it?

cabarnes, 2018-04-30:
Merged

closed 2018-04-30 by cabarnes