rovercode/rovercode #118: Use OAuth instead of user credentials to register rover

aninternetof, 2017-06-30:


Ryanb58, 2017-09-26:
This sounds like a good change. It would enable the open-closed S.E. principle so that if you guys decide to ever create a mobile app or something you'll be able to do so with ease.

aninternetof, 2017-12-17:
Like this, from [rovercode-web #103](https://github.com/rovercode/rovercode-web/pull/103):

**From the Rover, use credentials to obtain an access token**
```
$ http --verbose -f POST http://localhost:8000/oauth2/token/ grant_type=client_credentials client_id=ZXhNNFMBuLYiuLXEsGFHLh4XFo2mNMd5XFJHdWdU client_secret=HCYbSCC6aGZbiiaOYXAW3KWSb8ue2JO1Cqa1VB3EXLhklLn9SGCBTPYYMqzpaTkDUhevTgTdSnPJkoSwPqnLuQxXkv2DsWz8hm7dbtqO4aDDop4LitGMzw2h0PCMxd9G
POST /oauth2/token/ HTTP/1.1
Accept: */*
Accept-Encoding: gzip, deflate
Connection: keep-alive
Content-Length: 223
Content-Type: application/x-www-form-urlencoded; charset=utf-8
Host: localhost:8000
User-Agent: HTTPie/0.9.2

grant_type=client_credentials&client_id=ZXhNNFMBuLYiuLXEsGFHLh4XFo2mNMd5XFJHdWdU&client_secret=HCYbSCC6aGZbiiaOYXAW3KWSb8ue2JO1Cqa1VB3EXLhklLn9SGCBTPYYMqzpaTkDUhevTgTdSnPJkoSwPqnLuQxXkv2DsWz8hm7dbtqO4aDDop4LitGMzw2h0PCMxd9G

HTTP/1.0 200 OK
Cache-Control: no-store
Connection: close
Content-Type: application/json
Date: Sun, 10 Sep 2017 21:20:27 GMT
Pragma: no-cache
Server: Werkzeug/0.11.15 Python/3.5.4
Vary: Authorization
X-Frame-Options: SAMEORIGIN

{
    "access_token": "cSVmBlCfD9aAWMSlbVPBlpy5nAiEqg", 
    "expires_in": 36000, 
    "scope": "read write", 
    "token_type": "Bearer"
}
```
**From the Rover, make API calls using the access token**
```
$ http --verbose http://localhost:8000/api/v1/rovers/ Authorization:"Bearer cSVmBlCfD9aAWMSlbVPBlpy5nAiEqg"
GET /api/v1/rovers/ HTTP/1.1
Accept: */*
Accept-Encoding: gzip, deflate
Authorization: Bearer cSVmBlCfD9aAWMSlbVPBlpy5nAiEqg
Connection: keep-alive
Host: localhost:8000
User-Agent: HTTPie/0.9.2

HTTP/1.0 200 OK
Allow: GET, POST, OPTIONS
Connection: close
Content-Type: application/json
Date: Sun, 10 Sep 2017 21:22:30 GMT
Server: Werkzeug/0.11.15 Python/3.5.4
Vary: Accept, Authorization, Cookie
X-Frame-Options: SAMEORIGIN

[
    {
        "id": 2, 
        "last_checkin": "2017-09-10T21:22:27.698442Z", 
        "left_backward_pin": "XIO-P1", 
        "left_eye_pin": "XIO-P2", 
        "left_forward_pin": "XIO-P0", 
        "local_ip": "8.8.8.8", 
        "name": "Sparky", 
        "owner": 2, 
        "right_backward_pin": "XIO-P7", 
        "right_eye_pin": "XIO-P4", 
        "right_forward_pin": "XIO-P6"
    }
]
```
**An invalid token is blocked**
```
$ http --verbose http://localhost:8000/api/v1/rovers/ Authorization:"Bearer abc"                           
GET /api/v1/rovers/ HTTP/1.1
Accept: */*
Accept-Encoding: gzip, deflate
Authorization: Bearer abc
Connection: keep-alive
Host: localhost:8000
User-Agent: HTTPie/0.9.2

HTTP/1.0 403 Forbidden
Allow: GET, POST, OPTIONS
Connection: close
Content-Type: application/json
Date: Sun, 10 Sep 2017 21:24:43 GMT
Server: Werkzeug/0.11.15 Python/3.5.4
Vary: Accept, Authorization, Cookie
X-Frame-Options: SAMEORIGIN

{
    "detail": "Authentication credentials were not provided."
}
```

aninternetof, 2018-05-28:
Done.

closed 2018-05-28 by aninternetof