rovercode/rovercode #124: Can Bypass File Name Creation.

Ryanb58, 2017-09-11:
On `beta.rovercode.com` I am able to bypass the file naming step via these steps:

Navigate to "Live Demo".
You should see a "Name your design" modal.

Just click on the grey part outside of the modal and it goes away and then you can create a block but no longer save it.

aninternetof, 2017-09-11:
This is a good thing to point out. I think previously in this case, we gave the design a random name and saved it under that. Is that still the case?

Of course, why not just force people to name their designs? We could make that modal non-exitable.

What do you think is best?

Ryanb58, 2017-09-26:
From a users perspective, I think it would be best to force naming of a document. Else you'll just have a lot of randomly named partially used documents from people who want to create something but never do. Forcing them to name it is a cognitive instruction that they are going to accomplish the topic or task defined in the name. 

But that is just my personal opinion.

aninternetof, 2017-09-26:
I'm on board! Should be a very easy change, since I think Bootstrap modals have a "nonclosable" option.