rovercode/rovercode #128: Make standalone auth client

aninternetof, 2018-01-26:
Probably the best answer would be to make a client class that you could call get(), post(), etc on that would handle the authentication within. We could even make it a standalone project that this project imports (and anyone else who would want to interact with our API). But, for now, this is a good option.

cabarnes, 2018-01-28:
@aninternetof Looks like the answer is to use the `Backend Application Flow` in [requests-oauthlib](https://requests-oauthlib.readthedocs.io/en/latest/oauth2_workflow.html#backend-application-flow).

```python
In [1]: from oauthlib.oauth2 import BackendApplicationClient

In [2]: from requests_oauthlib import OAuth2Session

In [3]: client_id = '<id>'

In [4]: client_secret = '<secret>'

In [5]: client = BackendApplicationClient(client_id=client_id)

In [6]: session = OAuth2Session(client=client)

In [7]: session.fetch_token(token_url='https://beta.rovercode.com/oauth2/token/', client_id=client_id, client_secret=client_secret)
Out[7]: 
{'access_token': '<token>',
 'expires_at': 1517144236.0421565,
 'expires_in': 36000,
 'scope': ['read', 'write'],
 'token_type': 'Bearer'}

In [8]: session.get('https://beta.rovercode.com/api/v1/block-diagrams/')
Out[8]: <Response [200]>
```

It also handles refreshing the access token automatically: https://requests-oauthlib.readthedocs.io/en/latest/oauth2_workflow.html#third-recommended-define-automatic-token-refresh-and-update

aninternetof, 2018-01-28:
Oooh. Very cool. Let's definitely do this.

It had occurred to me that we currently have no means to refresh the token other than a reboot. So this will solve that problem.

Luckily for the meantime, 36000 is a pretty long time.