rovercode/rovercode #129: Query by client id not working

aninternetof, 2018-04-01:
When the rover first tries to get its own ID, it uses `?client_id=XXXX`. For some reason, this returns a list of **all rovers**. Right now, it ends up with a random one from that last.

This is related to #128 (and we should still do #128), but I think this problem would still remain after that change.

My thought is that we include the rover name in the .env, so that the rover can query for its ID by that. That will be a quick change. Any other suggestions?

cabarnes, 2018-04-01:
It should only be returning the list of rovers owned by the user: https://github.com/rovercode/rovercode-web/blob/development/api/views.py#L19

But, that still doesn't fix this problem. It doesn't filter by `client_id` because that isn't a filter field: https://github.com/rovercode/rovercode-web/blob/development/api/views.py#L15

We should be able to change the API to lookup through the `oauth_application` reference on each rover to add that filter. We could also create a REST API for applications that would only show your applications and then you could match the particular application with a rover.

aninternetof, 2018-04-01:
Oh, you' right, I misspoke: it is only showing the the rovers for the user. That would be a much bigger problem if it were returning *all* rovers. :-)

I believe the reference to the oauth application is on the rover model, and association of application to rover is already done at rover creation. So does that make it simple to add application.client_id as a filter?

I might need some more explanation on your last sentence (new REST API for matching applications to rovers).

aninternetof, 2018-04-02:
A quick test showed that `filter_fields` did not like `oauth_application.client_id` or `client_id` (how it's added to the serializer).

So I think we'll need to create a custom `FilterSet`.

cabarnes, 2018-04-02:
I believe it's `oauth_application__client_id` in DRF for a foreign key reference

aninternetof, 2018-04-02:
Bingo, that work perfectly! Thanks!

I'll put up the PR for both sides after work.

aninternetof, 2018-04-02:
Ah, ok, I see an example of that in the docs here: http://www.django-rest-framework.org/api-guide/relations/

closed 2018-04-06 by aninternetof