rovercode/rovercode #157: runtime error

zhangyanrui, 2018-12-26:
when I run "sudo bash start.sh ", I got an error like this:

 rovercode service starting
Traceback (most recent call last):
  File "app.py", line 25, in <module>
    pwm = pwmLib.get_platform_pwm(pwmtype="softpwm")
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/PWM.py", line 182, in get_platform_pwm
TypeError: __init__() got an unexpected keyword argument 'pwmtype'
rovercode service stopped

zhangyanrui, 2018-12-26:
I am not good at python and pip,  I don't know why I got this error!

aninternetof, 2018-12-26:
Hi zhangyanrui, thanks for writing this ticket! That's an interesting error -- we'll figure out what's happening.

One first question: Are you running the service on your laptop or desktop, on a CHIP, or on some other type of computer?

zhangyanrui, 2018-12-30:
I am running this in My RaspberryPi.

aninternetof, 2018-12-30:
Thanks!

Ok, I think I see what our issue is. On line 25 of www/app.py, we do this:

`pwm = pwmLib.get_platform_pwm(pwmtype="softpwm")`

I see now that the `pwmtype"softpwm"` only works on the CHIP hardware, not the Raspberry Pi. This is a mistake on our part.

Could you try editing that line so that it looks like this?

`pwm = pwmLib.get_platform_pwm()`

And then if you run it again, let me know if that error goes away!

aninternetof, 2018-12-30:
After that, we'll want to make sure that you set the names of the pins you have the motors hooked up to when you register the Rover on rovercode.com. If that gives you any trouble, we can talk through that here, too!

[image]